ronanguilloux/php-gpio #18: Raspebrry Pi A+ & version 2 support

ronanguilloux, 2015-03-20:
I'm putting here an email from Jack **** about Raspebrry Pi A+ & version 2 support:

>Dear Ronan Guilloux
>At First Thanks For make the Project php-gpio and share with us.
>I have a Question may i know is this Project Support the Raspberry-Pi A+ or Raspberry-Pi 2 ?
>I have Test this in My Raspberry Pi A+ with a simple LED blink but fail.
>So just want to make sure for this.
>Thanks 

My answer, and the things to do are:

>Hello Jack ****!
>AFAIK, A+ & 2 version are not supported yet, but it wouldn't be a big deal to add such support, since you just have to find the right list of GPIO pins to list in the Gpio class constructor.
>Such pins list for both A+ & Raspi 2 may probably be found in the Raspberry Pi public specifications. [Such (unofficial) spec is already online](http://elinux.org/RPi_Low-level_peripherals#GPIO_hardware_hacking), but a deeper check on that, from you or someone else would be nice.
>So, do not hesitate to contribute & to enhance [our GPIO pins list](https://github.com/ronanguilloux/php-gpio/blob/master/src/PhpGpio/Gpio.php), in order to support these new Raspi models !

Tjoosten, 2015-04-07:
If i get the sys info (Raspberry A+) i can start writing on this.

ronanguilloux, 2015-04-08:
Feel free to propose a PR: Official Raspberry Pi A+ hardware documentation [is here](https://github.com/raspberrypi/documentation)