ronanguilloux/php-gpio #25: Support for Raspberry Pi 3 Model B

winternet-studio, 2016-12-06:
For my Raspberry Pi 3 it doesn't detect the right version number in Pi->getVersion(). It returns 0 instead of 4. Therefore I can't use certain pins without it complaining. I would have made a pull request if I could but I don't know how to convert the revision numbers (http://www.raspberrypi-spy.co.uk/2012/09/checking-your-raspberry-pi-board-version/)

calcinai, 2016-12-06:
Hey, I've just tested a couple of things on my Pi and seems to be ok.  Can you attach the contents of your `/proc/cpuinfo`?

winternet-studio, 2016-12-06:
Of course...

```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2709
Revision        : a22082
Serial          : 0000000028d29c51
```

winternet-studio, 2016-12-06:
The Pi 3 by the way also has quite an extra number of GPIO pins: 5, 6, 12, 13, 16, 19, 20, 21, 26 (source: http://raspi.tv/wp-content/uploads/2014/07/Raspberry-Pi-GPIO-pinouts-1024x703.png)

calcinai, 2016-12-06:
Bad news for you is, I've tested with that cpu info file and I get `10625154` from `->getVersion()`.  Meaningless, but it should be working for you.

I'm not the maintainer of this library - I was just going to make the PR for you.

If you get desperate, you could try [my one](https://github.com/calcinai/phpi) in the meantime.

geofernz, 2017-04-24:
I have the same issue as @winternet-studio . On a RasPi running Raspbian Jessie, version detection is failing. Works fine when I manually edit the pins array. Maybe we can make the $pins property public so we can do an array_push on it as fallback? Here's my /proc/cpuinfo
`processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2709
Revision        : a22082
Serial          : 0000000009867659`

ronanguilloux, 2017-04-26:
Sorry, not time left nor budget to try this so far,
Please feel free to share a PR here